Zeugwerk/Twinpack #51: there should be a possibility to ensure if a library comes from a specific publisher

seehma, 2023-08-07:
maybe something like a hash or through a signing mechanism.

for example:
twinson project creator wants to publish twinson on this platform -> he publishes a public key on his website with which everyone can check if the published library is really created by him with his private key!?

only an idea...

iadonkey, 2023-08-07:
This means we have to verify the identity of distributors. We could do that by having distributors generate a key pair and let them upload their public key to the Twinpack Server. I think we should postpone this and make it optional later on to have have verified packages

iadonkey, 2023-12-08:
this is in contrast to twinpack registry, won’t do

closed 2023-12-08 by iadonkey