Zeugwerk/Twinpack #56: when adding zcore as placeholder, depencies are not shown correctly in publish dialogue

seehma, 2023-08-07:
if i add a dependency (e.g. ZCore) with a fixed version number it gets showed as follows:
[image]
[image]

When i add it as a placeholder like this
[image]

No dependencies are shown in publish dialogue
[image]

seehma, 2023-08-07:
but this is only done when i add a library as placeholder without auto resolution. when i select a version (for the placeholder), die Dependency is shown. Even when i do this outside of the catalogue

iadonkey, 2023-08-12:
ok, got it, I am not picking up references that are using the DefaultResolution

commit 6567ca48 (2023-08-12): fix: pick up references that use the default resolution
This closes #56
closed 2023-08-12 by iadonkey