Zeugwerk/Twinpack #58: modifying a published library changes the icon

seehma, 2023-08-07:
if we want to modify some library preferences but we do not change the library icon. the icon gets changed (in my case to a default icon of zeugwerk)

- so i open a library project with the beckhoff shell
- the right icon is shown on the modify dialogue

[image]

- now change some details, but leave the icon as it was 

[image]

- click on modify

[image]

- icon changed

iadonkey, 2023-08-07:
This is a backend issue

iadonkey, 2023-08-11:
fixed

closed 2023-08-11 by iadonkey