Zeugwerk/Twinpack #71: Unable to register to publish packages

Roald87, 2023-08-25:
I installed Twinpack in TwinCAT XAE. During the installation, I filled in my email address, but I didn't get any email. Should I receive one?

Then I open a project that I want to publish. I get the prompt to login and fill in my email and a random password. This then fails and it asks me to reset my password. I confirm and it leads me to https://twinpack.dev/wp-login.php, but that is not working.

seehma, 2023-08-25:
ok, iam sorry for that hustle, can you look into your spam folder? 
if there is not any registering email, can you try to register here -> https://zeugwerk.dev/wp-login.php

Roald87, 2023-08-25:
Thanks the linked worked. There was no email in my spam folder, I guess I made a typo when I installed Twinpack

closed 2023-08-25 by Roald87

seehma, 2023-08-25:
if you look into the Twinpack.iss file there is a function called ValidateEmail(strEmail).
Maybe there was a typo or your email address does not match the validation?

Roald87, 2023-08-25:
where can I find this file?

seehma, 2023-08-25:
one more question...
did you press the register button?

Roald87, 2023-08-25:
Not sure, but I guess so?

seehma, 2023-08-25:
> […](#)

here it is...
https://github.com/Zeugwerk/Twinpack/blob/main/Twinpack.iss

i think we will change the registering process during install to make it more userfriendly.

Roald87, 2023-08-25:
I didn't clone the repo, I just downloaded the exe and installed it.

When typing in the email, it would be good to at least have two fields, one
to repeat the email address.

seehma, 2023-08-26:
see -> https://github.com/Zeugwerk/Twinpack/issues/78