Zeugwerk/Twinpack #80: Add references not working if library title != library name

iadonkey, 2023-08-27:
If library title != library name there is an exception when trying to add the library, happens for the package "TwinCAT Dynamic Collection"

@fisothemes thanks for sharing you library with Twinpack. Unfortunately at the moment Twinpack has a this bug, but we will try to fix it

iadonkey, 2023-08-27:
Temporarily fixed the issue for fisothemes library, but the core issue remains, we need a seperate Title for every package, because the plcproj file can have a different name than the library

fisothemes, 2023-08-27:
I'll see if I can change the title on the project and upload an updated version. Thanks for opening Twinpack for everyone to use

iadonkey, 2023-08-27:
No need to change the title, it will be fixed with the next patch release, we can then have a title and a different name (see PR above)

PR #81 (2023-08-27): feat: support library title that is different from the library name
This closes #80
closed 2023-08-27 by iadonkey