Zeugwerk/Twinpack #86: Impossible to add library if project is encrypted

elconfa, 2023-09-01:
I found a problem: my project has software protection (encryption) of the project and Twinpack does not allow to import of any library into the project. By unencrypting the project, everything works fine.
Currently, to work around the problem, I unencrypt the project, add the library, and then put the encryption back before uploading the project to the PLC.

Thanks

[image]
[image]

iadonkey, 2023-09-01:
Can you please try the following:
- Unencrypt your project
- Create a config.json file by pressing the "Create" Button in the catalog.
- Encrypt your project again

Then you should be able to add libraries and keep the project encrypted

elconfa, 2023-09-01:
The "Create" button gives me an error (project is uncrypted)

[image]

iadonkey, 2023-09-01:
Sometimes the automation interface is a bit weird here. Can you to save and close the solution, open it again and press create again please?

elconfa, 2023-09-01:
I tried to unencrypt the project and close Twincat. Reopen the project, but the "Create" button still does not work.

iadonkey, 2023-09-01:
Unfortunately we can’t reproduce the problem. Could you upload the Twinpack Log, which is located in

%LOCALAPPDATA%\Zeugwerk\logs\Twinpack\Twinpack.debug.log

If possible, could you also upload your solution? We need the general structure of projects, plcs of the solution to debug this. Maybe you have non-plc projects in your solution? (This would cause an issue atm). Of course you can delete all the source code for this.

Since Twinpack is Open source you could also try to debug it yourself. We will write a guide how to debug the vsix, but it boils down to this:

- Clone the repository 
- Open Twinpack.sln with VS2022 and build the solution 
- For XaeShell, copy the Content of TwinpackVsix/bin/Debug/Package to C:/Program Files (x86)/Beckhoff/TcXaeShell/Common/Extensions/Zeugwerk/Twinpack (overwrite the files that the installer placed here)
- Open you Solution in TcXaeShell and navigate to the Twinpack Catalog
- In VS2022 click Debug -> Attach to process and select the TcXaeShell
- Place a breakpoint in the File Catalog

closed 2024-02-01 by iadonkey