Zeugwerk/Twinpack #87: my library is not published...

runtimevic, 2023-09-06:
Hello @seehma , @iadonkey , @zeugwerker 

My library is not published...
Is there a log site or where I can see why it is not published?
thank you, Víctor.

iadonkey, 2023-09-06:
Twinpack is only uploading libraries when check all objects is going through without errors. According to the log this is not the case here. 
In the output window of your Visual Studio / TcXaeShell there should be some indications why check all object fails. In order to upload the library you have to fix these errors.

However, it is actually a bug that there is no message box popping up to indicate that check all objects failed, we will fix this in the next patch.

runtimevic, 2023-09-06:
- I had to create this route by hand:
'E:\BECKHOFF_TcMotion\Component_Motion_OOP_Axis\TC3_Axis.Zeugwerk\libraries\TC3.1\
- I had to install the library and leave it in the indicated path:
'E:\BECKHOFF_TcMotion\Component_Motion_OOP_Axis\TC3_Axis.Zeugwerk\libraries\TC3.1\TcMotion_Library_1.0.0.9.library'.

Is this normal? Should these 2 steps be performed?

iadonkey, 2023-09-06:
No, and this is actually only possible due to the bug I mentioned.
The log message indicates that check all object for your PLC failed and that's why Twinpack wasn't creating the folder and saving to library to this location. The normal process requires that check all objects get through without any errors

runtimevic, 2023-09-06:
Hi, @iadonkey ,
Yes, this is the problem:

Message:		Compile complete -- 135 errors, 101 warnings

Thanks you, Víctor

iadonkey, 2023-09-06:
Do you want me to delete the package you uploaded from the Twinpack Server for now? In any case, if the errors are fixed you can simply upload a new version (i.e.1.0.0.10)

closed 2023-09-06 by iadonkey

iadonkey, 2023-09-06:
Sure, but I hope you understand that we will prevent the workaround for future releases 😉

commit 086d4954 (2023-09-06): fix: if check all objects fails there is no visual indication about it besides the log

this closes #87

(cherry picked from commit 9b4fe8f6e9b499dc709f40edc017fbefcc8dc006)