Zeugwerk/Twinpack #88: icon change libray  pixel size?

runtimevic, 2023-09-06:
Hello @seehma , @iadonkey , @zeugwerker :

The library display icon does not load, what should the icon be like so that it can be chosen?
[image]

Thanks you, Víctor.

iadonkey, 2023-09-06:
it should be a square png image (i.e. 256x256), if no icon is chose one is generated automatically with a hash.
Could you post the image that you tried so we can debug the issue?

runtimevic, 2023-09-06:
My image does not have the appropriate pixel resolution, I'm going to change it and try it.
thanks...

runtimevic, 2023-09-06:
Hi, 
I have changed the resolution of the icon and it still does not load..., when choosing it should it be displayed?
[image]

iadonkey, 2023-09-06:
Please move the icon into the same folder as the solution and try again, you don't have to publish a new version for this, you can also use "Modify Package" for this (Twinpack Package -> Modify ...)

It was previously a prerequite to have the icon in the same folder. This was later changed, but there seems to be a bug that it doesn't work as intended if the file is on a different drive than the solution

iadonkey, 2023-09-12:
@runtimevic did you try with moving the file to the folder of the solution? Alternatively, if you update to the latest version (0.2.15) it should work from any location.

runtimevic, 2023-09-12:
I have to try it...

closed 2024-02-01 by iadonkey