Zeugwerk/Twinpack #116: Availability of Beckhoff library

GISED-Link, 2024-02-01:
Hello,

Are the beckhoff libraries also available (Tc2_Math, Tc2_MC2_Drive, Tc2_Drive, Tc2_TcpIp, etc...) in your repo? It is a pain in the ass when you have toggled the 'set to effective version' parameter and someone's laptop is missing one particular version of a lib. Would be greate to have a tool that is capable of downloading the missing version of the lib directly from the internet. I don't know if there can be legal problem on such a thing since those lib are compiled libraries...

Thanks

iadonkey, 2024-02-01:
I feel your pain ... but at least to my understanding of Beckhoff EULA we are not allowed to distribute anything where they have the copyright on. That is why we didn't upload any of their packages. I don't believe Beckhoff will, but they could do it by themselves if they wanted.

**BUT...**
In TwinCAT 4026, Beckhoff will publish a package manager called tcpkg, which is based on [chocolatey](https://chocolatey.org/). For now this doesn't seem to be embedded into IDE, but we don't know their final solution so maybe this will change. We see two scenarios here:

1. The integrate neatly into the IDE so that XAE Shell will automatically offers some way to download and install the missing libraries when you open a PLC, but more likely ...
2. This will remain a commandline/external tool, which we can use in Twinpack to download and install the missing libraries. This would not be a copyright violation since their official package manager is doing all the work and Twinpack is just triggering the action.

We didn't start to work on this (https://github.com/Zeugwerk/Twinpack/issues/99), because we want to wait until their 4026 release to see where they are going - We want to avoid wasting resources on something, which Beckhoff's solution is already providing.

GISED-Link, 2024-02-01:
Ok thanks for the answer. Waiting on the 4026 was also my first approach. But since I don't know when it will comes out nor if the version will be stable enough, I was hoping to find a good solution in the mid-time.

closed 2024-02-01 by iadonkey