Zeugwerk/Twinpack #135: Unable to LogIn into Azure Artifact Server

ThatsFokus, 2024-07-29:
When adding a NuGet Source from an Azure Repository it seems to be impossible to Log In, Security Dialog just closes and open again wether the credentials are correct or not. Also besides killing the Shell with Taskmanager it is impossible to stop the dialog from opening again

iadonkey, 2024-07-29:
Can you post the detailed log messages from %AppData%\ Zeugwerk\logs\Twinpack\Twinpack.debug.log

The credentials are verified by checking the feed for packages, are there any in the feed? https://github.com/Zeugwerk/Twinpack/blob/264bb296cfad91ee64cffee77453d78483b5adbf/TwinpackShared/Protocol/Nuget/NugetServer.cs#L479

iadonkey, 2024-07-29:
> Also besides killing the Shell with Taskmanager it is impossible to stop the dialog from opening again

will be fixed, https://github.com/Zeugwerk/Twinpack/issues/136

ThatsFokus, 2024-07-30:
[Twinpack.debug.log](https://github.com/user-attachments/files/16423010/Twinpack.debug.log)
[Countdown.nuspec.txt](https://github.com/user-attachments/files/16423107/Countdown.nuspec.txt)

@iadonkey  Here is the Log and and a nuspec for one of the uploaded packages

iadonkey, 2024-07-30:
This is a parsing error coming from `NuGet.Packaging.Core`, are you sure that the URL for your feed is correct and returning a valid xml for NuGet? Is there something reasonable visible in the browser when you navigate to the feed's URL?

ThatsFokus, 2024-07-30:
checking the source with nugetCLI it has the package listed as avaiable, the same happenes in the web interface. Could there be other reasons why it has said Parsing error?

iadonkey, 2024-07-30:
could you post the xml it returns if you go to the feed? maybe try to disable authentication on the feed for testing? is there an API key, which is need to access the feed?

ThatsFokus, 2024-07-30:
Hey, managed to connect the feed after some more trouble Shooting including manually editing the index.json.

Sadly I am currently unable to see the Package in the Twincat Extensions feed. 

 

> could you post the xml it returns if you go to the feed? maybe try to disable authentication on the feed for testing? is there an API key, which is need to access the feed?

how should I go about reading said xml?

iadonkey, 2024-07-30:
If login now works then there is no need for doing anyhting with the xml. The exception you posted indicate that the operation for getting the packages was canceled, usually this happens when you press the "X" button while the query is running. Is there some different exception if you just let it run? is it timing out or anything like that?

iadonkey, 2024-07-30:
what did you change in the index.json bzw? there shouldn't be any need to changes this, we are using the official nuget package to connect to nuget servers, as long as version 3 is supported by the server everything should be fine. We don't have an azure instance to test, but it works with nexus and https://github.com/idoop/docker-nuget-server

ThatsFokus, 2024-07-30:
[image]
[image]
[image]

> If login now works then there is no need for doing anyhting with the xml. The exception you posted indicate that the operation for getting the packages was canceled, usually this happens when you [...]

Seemingly it just doesn't find anything, there isn't any exception but as shown in the first screenshot the package is published on the feed.

Should you be able to tell any problem with the nuget itself I would be happy about an idea

iadonkey, 2024-07-30:
if there is "Library" with a capital "L" in the tags, we had to include some heuristics for Beckhoff. Could you try to add an additional Tag after Library with the name of the library, so it looks like "Small CountDown Library CountDown"?
We accidently use the heuristics for the normal nuget servers as well, so this might be the problem

ThatsFokus, 2024-07-31:
[Countdown.nuspec.txt](https://github.com/user-attachments/files/16445053/Countdown.nuspec.txt)

> if there is "Library" with a capital "L" in the tags, we had to include some heuristics for Beckhoff. Could you try to add an additional Tag after Library with the name of the library, so it looks [...]

Hey sorry for coming back to you this late. As shared somewhere above here is my nuspec once again.

I didn't really follow what I would have to change so it conforms with Beckhoffs heuristics.

iadonkey, 2024-07-31:
Not sure where the „Small“ and „Library“ tag in your screenshot are coming from with this nuspec

anyway, we‘ll write some documentation how to publish a nuget package which is useable with Twinpack. Just a bit busy at the moment

iadonkey, 2024-07-31:
btw. this is what beckhoff nuspecs look like

```xml
<?xml version="1.0"?>
<package xmlns="http://schemas.microsoft.com/packaging/2011/08/nuspec.xsd">
  <metadata>
    <id>TwinCAT.XAE.PLC.Lib.Tc2_Math</id>
    <version>3.4.4</version>
    <title>Beckhoff TwinCAT XAE PLC Library Tc2_Math</title>
    <authors>Beckhoff Automation</authors>
    <owners>Beckhoff Automation</owners>
    <projectUrl>https://www.beckhoff.com/en-us/products/automation/twincat/</projectUrl>
    <requireLicenseAcceptance>false</requireLicenseAcceptance>
    <description>Contains Beckhoff TwinCAT XAE PLC Library Tc2_Math version 3.4.4.0</description>
    <summary>TwinCAT-XAE-PLC-Lib-Tc2_Math package</summary>
    <releaseNotes>Artifact Information
  TwinCAT-XAE-PLC-Lib-Tc2_Math 3.4.4+76e0a95d
  Setup 1.0.1.10+0b47908d</releaseNotes>
    <copyright>(c) Beckhoff Automation GmbH &amp; Co. KG</copyright>
    <tags>Beckhoff TwinCAT XAE PLC Library Tc2_Math Setup InstallationLevel1 AllowMultipleVersions</tags>
    <docsUrl>https://infosys.beckhoff.com/</docsUrl>
    <dependencies>
      <dependency id="TwinCAT.Prep" version="0.0.0" />
    </dependencies>
  </metadata>
</package>
```

using this and adapting to your needs should work as the NuGet Server implementation is basically the same as the Beckhoff Server implementation in Twinpack.

Just make sure that the Tag after "Library" (in <tags> and <title>) is the title of your library (as in your .library file) - this little annoyance will be removed in one of the next patches of Twinpack, since for .library files we can actually get the title directly from the file.

dfreiberger, 2024-08-02:
@ThatsFokus regarding logging in to Azure DevOps artifacts server, I was able to do so by using my email and a PAT (personal authentication token) which I generated in Azure DevOps. I did not modify any configuration files in Twinpack and just logged in when prompted.

I also ran into an issue after connecting where I couldn't see packages that I had published. After troubleshooting it looks like the Published field is not populated by Azure DevOps Artifact feeds, but is used in Twinpack.

I modified 
https://github.com/Zeugwerk/Twinpack/blob/main/TwinpackShared/Protocol/Nuget/NugetServer.cs#L122 to the below and the packages showed up.
```
                    new Tuple<IEnumerable<CatalogItemGetResponse>, bool>(
                        results.Select(x =>
                            new CatalogItemGetResponse()
                            {
                                PackageId = null,
                                Name = x.Identity.Id,
                                DistributorName = x.Authors,
                                // Description = x.Description, Beckhoff's descriptions are meh
                                IconUrl = x.IconUrl?.ToString() ?? IconUrl,
                                RuntimeLicense = 1,
                                DisplayName = x.Identity.Id,
                                Downloads = x.DownloadCount.HasValue ? (int)x.DownloadCount.Value : 0,  // Added null check
                                Created = x.Published?.ToString() ?? "Unknown",  // Added null check
                                Modified = x.Published?.ToString() ?? "Unknown"  // Added null check
                            }).ToList(),
                        results.Any());                     
```

iadonkey, 2024-08-02:
fixed with release 1.0.8

closed 2024-08-02 by iadonkey