Zeugwerk/Twinpack #143: Icon not displayed when deploying a library via Nuget

georgtauber-durst, 2024-11-04:
In combination with a NuGet-Server the Icon is not displayed.

The icon is integrated according to the official Nuget reference (https://learn.microsoft.com/en-us/nuget/reference/msbuild-targets#packing-an-icon-image-file)

e.g.

```
<PropertyGroup>
    ...
    <PackageIcon>icon.png</PackageIcon>
    ...
</PropertyGroup>

<ItemGroup>
    ...
    <None Include="images\icon.png" Pack="true" PackagePath="\"/>
    ...
</ItemGroup>
```

seehma, 2024-11-05:
Currently Twinpack only supports <iconUrl>. We have registered your request for the <icon> tag (as described in the nuspec documentation) as feature request and will reach out back to this issue when its implemented.

iadonkey, 2024-12-07:
Good news, starting with the next major release of Twinpack (version 1.2.x) the icon tag will be supported as it is documented in the official NuGet "nuspec reference"

[image]

Since this is not critical issue, the feature is not backported into Twinpack 1.0.x and we hope to do the release of Twinpack 1.2 in January 2025. Stay tuned and thank you for the feature request

closed 2024-12-07 by iadonkey