Zeugwerk/Twinpack #147: Issues when modifying a published package

ZigaJavornik, 2024-12-28:
Hi,

I am quite new to this platform, but I am a huge fan of sharing libraries and was very excited when I saw that Twinpack exists. I have an issue, where I want to modify my own released package. When I click the "edit" button, i get the following message:

[image]

I then opened Twinpack, and saw the image (yes, it is silly) was updated, but notes were not added, for example. Not sure if I can provide some more information, let me know. Also adding Twinpack console output

  INFO Modifying package
  INFO Updating general package
  INFO Updating package version
  INFO Updating package configuration <path_to_project>\.Zeugwerk\config.json

iadonkey, 2024-12-28:
- regarding the message box: In the advanced panel. Do you have „compiled library“ checked? We don’t support this for public repos, it‘s a feature for our customers
- Some of the fields are for the package as a whole (license, project url etc.), others for a specific release of this package (version, notes). So at the moment you can’t update the notes of a release with this gui. We can change the text for you if you want though, just send it here and we‘ll update it

In general, you might want to have a look at https://github.com/Zeugwerk/Twinpack-Registry , it makes releasing on Twinpack simpler by directly scanning the repository listed in https://github.com/Zeugwerk/Twinpack-Registry/blob/main/repositories.txt for new (GitHub) releases: if your repo has a GitHub release, exactly one .library file in the assets with a project version not yet available on Twinpack, a [GitHub action](https://github.com/Zeugwerk/Twinpack/actions) automatically pushes the release to Twinpack.

iadonkey, 2024-12-28:
Just saw that your [release](https://github.com/ZigaJavornik/ZigaLibs/releases/tag/v0.0.1) contains several .library files. I am not sure if Twinpack Registry is not supporting this at the moment, but if not, I‘d test/implement it if you added your repo to Twinpack Registry

ZigaJavornik, 2024-12-29:
This makes sense then, just the single window with the error messages makes the user believe that modification failed. If you can, modify the "notes" to :
"You can find an example project at: https://github.com/ZigaJavornik/ZigaLibs/tree/master/Examples/Logging"

I can't seem to see a check box for compiled library in the advanced tab. There is only "type" of library, and it is set to "library" as opposed to "compiled library", which is the other option. I think this might be a bug - the field is not editable, there were no errors when I published it.
[image]

As far as automatic releasing goes, I keep floating between the idea of having 1 repo with multiple libraries, or split repositories with individual library releases... Should I swap to the latter, I will set up the automatic scanning and releasing from the released GitHub file. 

Thanks for the prompt reply.

iadonkey, 2024-12-29:
I’ll look into this bug

For our own libraries, we have a repo for every library and a “framework” repo, which has the library repos as submodules. The framework repos also contains examples, documentation (the non-api-reference one), installer scripts etc. this works quite nicely if you have CI to automate the overhead you get by doing this

ZigaJavornik, 2024-12-29:
> I’ll look into this bug
> 
> For our own libraries (e.g. Zeugwerk Framework libraries, which consist of ZCore, ZPlatform, ZAux, ZEquipment, ZApplication), we have a repo for every library and a “framework” repo, which has the library repos as submodules. The framework repos also contains examples, documentation (the non-api-reference one), installer scripts etc. this works quite nicely if you have CI to automate the overhead you get by doing this

I did look into submodules, and my commit history for the last day has been "add submodule->revert->add..." haha. I will take a deeper look perhaps, thanks.

iadonkey, 2024-12-29:
@ZigaJavornik, could you try again please to update the note via the user interface? There was a regression on the server-side due to adding initial support for filetypes different than (.library and .compiled-library), which will be needed for Beckhoff's PLC++

Fixed with https://github.com/Zeugwerk/Twinpack-Server/commit/589912938a8fd6456f12503d77494692ba45b3fa

Back to Twinpack Registry: Your current structure with multiple .library files per release would be supported already, so feel free to do a pull request if you want to have GitHub releases automatically on Twinpack

ZigaJavornik, 2024-12-29:
@iadonkey  If I understand correctly, there is no Twinpack update for this, since it is on the server side? If that is the case, I tried and still can't edit the "notes" field.

[image]

As for the "prefix" of the libraries, I will add it in the project it-self, I edited the names for all 3 library names in the Twinpack GUI, it still popped up the message "compiled" not assigned, though.

iadonkey, 2024-12-29:
Sorry about that, deploying the server patch had an issue, now it should actually work (no client updated needed)

ZigaJavornik, 2024-12-29:
> Sorry about that, deploying the server patch had an issue, now it should actually work (no client updated needed)

Indeed it does, we can close this now :)

closed 2024-12-29 by ZigaJavornik