Zeugwerk/Twinpack #149: Library is only displayed once if it is available in 2 repositories

georgtauber-durst, 2025-02-04:
If the same library is avialabel in 2 repositories (nuget) for testing reasons (stabel + Testing) the library is shown only once.

Both repos are connected and login into TwinPack (Package Server List)

Current workaround: logout from the "stable" repo to access the "testing" and vice versa

iadonkey, 2025-02-04:
This is per design, the repository with the higher priority (further up in the list), wins. It is the same behavior as NuGet uses

iadonkey, 2025-02-04:
Having different configurations/branches for the same package is a feature of Twinpack Repositories. NuGet does not natively support this.

iadonkey, 2025-02-12:
… On the right top corner, next to the configure button there is a combobox, which allows for filter packages for a certain repository. Did you try to select the one with the lower priority? This should lead to having the package pulled from the selected one. This is simpler than your current workaround

closed 2025-02-14 by seehma