Zeugwerk/Twinpack #151: Library (Nuget repository) is not shown in TwinPACK if the name is longer than 24 character

georgtauber-durst, 2025-03-07:
Apparently TwinPACK cannot handle library names longer than 24 characters.

If a library (e.g. ‘L_LEAIAC_IndexAccessClient’ = 26 characters) is generated with the following nuspec and stored on Nuget Server, it will not be displayed in twinPACK.

```
<?xml version="1.0" encoding="utf-8"?>
<package xmlns="http://schemas.microsoft.com/packaging/2013/05/nuspec.xsd">
<metadata>
    <id>L_LEAIAC_IndexAccessClient</id>
    <version>1.0.0.0</version>
    <title>L_LEAIAC_IndexAccessClient</title>
    <authors>MyCompany</authors>
    <projectUrl>https://bitbucket.org/MyCompany/L_LEAIAC_IndexAccessClient</projectUrl>
    <description>MyPlcLibrary is an awesome library to do ...</description>
    <copyright>MyCompany</copyright>
    <tags>library</tags>
</metadata>
<files>
    <file src="l_leaiac_indexaccessclient.compiled-library" target="TC3.1" />
</files>
</package>

```

If the name of L_LEAIAC_IndexAccessClient (26) is reduced to L_LEAIAC_IndexAccessClie (24) in <id> and <title>, the library is displayed correctly

```
<?xml version="1.0" encoding="utf-8"?>
<package xmlns="http://schemas.microsoft.com/packaging/2013/05/nuspec.xsd">
<metadata>
    <id>L_LEAIAC_IndexAccessClie</id>
    <version>1.0.0.0</version>
    <title>L_LEAIAC_IndexAccessClie</title>
    <authors>MyCompany</authors>
    <projectUrl>https://bitbucket.org/MyCompany/L_LEAIAC_IndexAccessClient</projectUrl>
    <description>MyPlcLibrary is an awesome library to do ...</description>
    <copyright>MyCompany</copyright>
    <tags>library</tags>
</metadata>
<files>
    <file src="l_leaiac_indexaccessclient.compiled-library" target="TC3.1" />
</files>
</package>

```

iadonkey, 2025-03-07:
is it not shown at all in the Catalog with the id being 26 characters? nuspec in theory limits the id to 128 characters, but this might be handled differently by some NuGet servers.

What server are you using and what version of NuGet protocol is it implementing? How does the library look like in the NuGet json feed when it has > 26 characters? A log might also be useful

As you see here https://github.com/Zeugwerk/Twinpack/blob/4646d1d0e484baaea017e431ad992434a9256bdb/TwinpackCore/Protocol/Nuget/NugetServer.cs#L102 , Twinpack is not limiting the Id in any way, it simply requests the feed with the official nuget package

seehma, 2025-03-26:
fixed after setting the necessary tag library in nuspec file.

closed 2025-03-26 by seehma