Zeugwerk/Twinpack #152: Extension not loading

iadonkey, 2025-03-18:
After installation it doesn't work for me anymore

[image]

iadonkey, 2025-03-18:
Can you please send the xml file mentioned in this exception?

Please also delete these folders and try again. Maybe you are coming from an old installation and files are not cleaned up

```

C:\Program Files (x86)\Beckhoff\TcXaeShell\Common7\IDE\Extensions\

C:\Program Files\Beckhoff\TcXaeShell\Common7\IDE\Extensions\

```

elconfa, 2025-03-19:
Good morning,

Before deleting this folders:
[ActivityLog.xml.txt](https://github.com/user-attachments/files/19334908/ActivityLog.xml.txt)

After deleting this folders:
[ActivityLog after.xml.txt](https://github.com/user-attachments/files/19334937/ActivityLog.after.xml.txt)

The problem still be present.

Thanks

iadonkey, 2025-03-19:
Good catch thank you.

As a fast solution, please create a file here

```
%appdata%\Zeugwerk\Twinpack\sourceRepositories.json
```

with this content:

```json
{
  "repositories": [
    {
      "Name": "twinpack.dev",
      "Url": "https://twinpack.dev",
      "ServerType": "Twinpack Repository"
    },
    {
      "Name": "public.tcpkg.beckhoff-cloud.com (stable)",
      "Url": "https://public.tcpkg.beckhoff-cloud.com/api/v1/feeds/stable",
      "ServerType": "Beckhoff Repository"
    }
  ]
}
```

We will build a patch release with a fix

elconfa, 2025-03-19:
Thanks, I did as suggested. 

Now the error message no longer appears, but the Twinpack entry no longer appears either.

[image]

I tried to relaunch the installation, but it is not there yet.
I completely uninstalled Twinpack and then reinstalled version 1.2.2 and now the error has reappeared.

I await the new version with the pach.

Thank you

iadonkey, 2025-03-19:
Can you send the xml again please? Maybe you deactivated it popping up?

iadonkey, 2025-03-19:
Meanwhile, here the fix for the missing sourceRepositories.json https://github.com/Zeugwerk/Twinpack/releases/tag/1.2.3.0

seehma, 2025-03-19:
Ok, fine! 
Can you check if you have this menu-entry?

[image]

Some nodes in Twincat will be added differently in different languages. We are mainly testing with english as set language and had problems when we started with Twinpack, when the language setting was set to a different one than english (e.g. german).

closed 2025-03-19 by iadonkey