Zeugwerk/Twinpack #154: connection refused with a Nuget Repository with empty credentials

philipphagen87, 2025-06-26:
It was not possible for me to connect to a NuGet Repository with empty credentials. After inserting the Name and URL i pressed login.

[image]

[image]

in the output window it was logged "Successfull", chain icon was connected I closed the window with "ok" and the new server was not shown on the repo list beside the configure Button and no packages was found.
So I go back to configure dialog and connection chain icon was disconnected and button showed "login" again.

[image]

In my case there are no credentials neccasary. I also tested with insert credentials -> than it works great.

iadonkey, 2025-06-26:
Thanks for reporting the issue, I‘ll look into it. As a workaround it should be ok to simply put any username and password (?)

iadonkey, 2025-06-26:
Fixed with a7331dc6f633f7d55e79facb1a2765e87759191e

iadonkey, 2025-06-27:
You don’t have to press the login button. For servers without credentials just put in the url and close the configuration dialog

iadonkey, 2025-07-04:
can we close this issue or is it not possible to browser packages on the repository without being logged in?

philipphagen87, 2025-07-05:
it works.

closed 2025-07-05 by philipphagen87